thorlabs-tsi/ThorDAQ_Issues #11: Channel A - No image, no interrupt on older Card, works on newer Card in VA.

rbelanger-thorlabs, 2018-03-07:
Channel A & 1 are the same.

rbelanger-thorlabs, 2018-03-07:
Bill R:
Took a quick look at the implemented design and saw that the timing path "to" the internal control registers was included in the set_false_path constraints.  The "-through" is should be "-from" when using "cells" as the target.  Enumeration using -through will include any clock used as a source for these cells.

That means the register writes from the "Target_Interface" (NWL DMA core) were not constrained.

zhunshi01, 2018-03-29:
There are 3 issues caused image acquisition to fail occasionally. 
1. Synthesis Strategy
2. There's a clock from MMCM in Sampling_Clock_Gen that has a wrong phase setting.
3. Toggling between external and internal timing could stop image acquisition. Manually switching JESD sync settings in GUI fixed the issue. This is a SW issue.

jmills-thorlabs, 2018-04-04:
If all goes good wrapped up to day

zhunshi01, 2018-04-06:
Apply 3 FPGA fix and a SW change for switching JESD settings should fix this bug:

1) Set Clkout1 Phase Offset to 70 in Sampling Clock Gen IP and regenerate the IP.
2) Use a latest timing constraint file from VA that just removes unused SCS Mezzanine port signals & constraints.
3) Set synthesis strategy to Vivado_Synthesis_Defaults