thorlabs-tsi/ThorDAQ_Issues #8: External clock counter returns 0 for 80 MHZ test.

rbelanger-thorlabs, 2018-03-07:


jmills-thorlabs, 2018-03-29:
This needs to be tested to make sure it is still not working. This may have been resolved with another fix.

jmills-thorlabs, 2018-04-02:
Leo will retest today. He was still seeing this in January.

rbelanger-thorlabs, 2018-04-06:
Correct Operation + ThorImage configuration was provided by Zhun
Correct Operation through FPGA register reads with arbitrary Laser Clock input was provided by myself.

Common thought is there may be an issue with Leo's hardware and/or configuration. More testing will be done by Leo on his setup.

[image]
[image]

LMandujano-thorlabs, 2018-04-06:
This feature seems to be working correctly now. Zhun and Ryan retested on their respective boards and did not see the issue. Closing the issue.

closed 2018-04-06 by LMandujano-thorlabs