thorlabs-tsi/ThorDAQ_Issues #13: Inconsistent Flash Operation - Boot + Normal Image

rbelanger-thorlabs, 2018-03-08:


jmills-thorlabs, 2019-01-02:
this may be related to the update tool out of sync. Baldwin is working on that tool. This may be something we still need to look into this.

ASiddiqui-thorlabs, 2019-02-22:
We need more description of the issue to reproduce it.
We tried TSI Firmware Update tool, 'Leo tool(un-controlled source control version)' for golden and production images. And we did not see any relevant issue.
1)Testing with  'Old VA Firmware Update tool' (un-controlled source control version)
Golden:  Not working at TSI
Production:  Not working at TSI
2)Testing with 'new VA tool'(un-controlled source control version):
Golden: N/A
Production: Working at TSI.