thorlabs-tsi/ThorDAQ_Issues #22: Variable gain for detectors not working

jmills-thorlabs, 2018-08-01:


elieser-thorlabs, 2018-08-13:
Configuring the front end attentuator is hard coded in 1.0.2d.  The control register should be re-enabled:

_ADC Interface Subsystem:  ADC_DAT_12: BAR 3, Base_Addr + 0x08, Write-only, 8-bit register with 4-bits for each DAT and 1-bit for AFE Shutdown.

Two channels are configured by this register, with 4-bits assigned to each channel.  Bits 3..0 control channel 1, bits 7..4 control channel 2.  The 3 LSBs of each nibble select the attenuation.  The MSB controls the AFE shutdown, with 0  Shutdown/OFF and 1 Enabled/ON.

The following settings determine what front-end attenuation is selected: 
    000b = 1dB
    001b = 5dB
    010b = 9dB
    011b = 13dB
    100b = 17dB
    001b = 21dB
    110b = 25dB
    111b = 29dB

Example:  Set channel 1 to 17dB and enable its AFE, and set channel 2 to 1dB and enable its AFE…
    ADC_DAT_12 = 0x9C  ( 1001b – channel 2,  1100b – channel 1 )_

BNNISERadtke, 2019-01-02:
This feature has been re-enabled in Firmware called "3p_ClkRx" given to Ashraf (we need an official version # to reference).  The relevant FPGA code is in "ThorDAQ_ADC.v" in the Vivado project, with the signals "fmc_dat12_config", "fmc_dat34_config", "fmc_dat56_config" in the module: "i_ADC_FMC_IF_Subsystem_Registers".   

Enabling this feature at the user level can cause great confusion and misuse of the analog front-end RF gain.  It is advisable to keep this hidden as an expert-user only selection.