thorlabs-tsi/ThorDAQ_Issues #30: Switching detector polarities in the GUI has no effect on the image until imaging is stopped and started again.

bgbritton, 2019-01-04:
This is not consistent with the other Galvo Galvo and Galvo-Resonant modules we have.
The ThorDAQ software should be able to take the new polarity and flawlessly update the LUT so the new outgoing images have the correct intensity data.

elieser-thorlabs, 2019-01-04:
Or we could just stop the imaging when this control is toggled.  This is very much an initial configuration setting, not something a user will be changing, ever.

bgbritton, 2019-01-04:
I agree. ThorImage will stop and start automatically if it is a change like dwell time, pixel density. It just not currently one of those settings right not in how the GG and GR dlls are setup. Should be an easy fix.

jmills-thorlabs, 2019-01-09:
Should there be a dialogue to tell the user that the polarity of the detector is wrong. It can be confusing.