thorlabs-tsi/ThorDAQ_Issues #46: I2C address change on new board caused communication with CPLD to fail

zhunshi01, 2019-01-25:
Tried CPLD firmware update on older main boards, works great.
Tried CPLD firmware update on new main boards, failed.

Which means there's hardware change on the main board that VAR was not acknowledged of.

Talked with Leo on Feb last year but what he said was not clear. Emails attached.

TSI, please verify the board change with VAR and work with us to get this issue fixed. Most likely some address change in software could resolve this issue.

[email032818.txt](https://github.com/thorlabs-tsi/ThorDAQ_Issues/files/2796715/email032818.txt)
[email022918.txt](https://github.com/thorlabs-tsi/ThorDAQ_Issues/files/2796716/email022918.txt)

zhunshi01, 2019-02-08:
On old boards, resistors R87, R88, R89 are installed.
On new boards, resistors R95, R94, R89 are installed.

Changed the PCA9548A address from 0x77 to 0x71 in software enabled the CPLD firmware update function to work again.

jmills-thorlabs, 2019-02-13:
this was fixed by ashraf and tested by zhun.

closed 2019-02-13 by jmills-thorlabs